Infineon/arduino-pas-co2-sensor #4: Forced compensation

Lu-Ma-Ni, 2022-02-27:
Hello all,
thank you very much for providing the library.

I am not understanding how with the latest version a calibration is done with a known CO2 concentration (Forced compensation). Can someone explain me how this is done. 

Thank you!

jaenrig-ifx, 2022-02-28:
Hi @Lu-Ma-Ni,

Please, find out more information in this [application note](https://www.infineon.com/dgdl/Infineon-AN_FCS_ABOC_XENSIV_PASCO2_2-ApplicationNotes-v01_02-EN.pdf?fileId=5546d4627b830c27017b8f0a98ad6354).

The specific API function to set the ABOC parameters is this one: https://github.com/Infineon/arduino-pas-co2-sensor/blob/bcd5899b9766797657f1f7ba675d8cf18d7f73d7/src/pas-co2-serial-ino.hpp#L42****

jaenrig-ifx, 2022-04-13:
Hi @Lu-Ma-Ni,

Did that information resolved your query?

closed 2022-04-29 by jaenrig-ifx