Infineon/arduino-pas-co2-sensor #5: Get new values only every 60 seconds

Lu-Ma-Ni, 2022-02-28:
**Prerequisites**
Can you reproduce the problem reliably?
Yes

**Description**:
I only get a value transmitted every 60 seconds, even though the period is set to 7 seconds, for example. If interprupts are used, you can see exactly that it is always 60 seconds. Even a reset does not solve the problem. 

**Steps to Reproduce:**
1. Upload the [serial-peridoic](https://github.com/Infineon/arduino-pas-co2-sensor/blob/bcd5899b9766797657f1f7ba675d8cf18d7f73d7/examples/serial-periodic/serial-periodic.ino) Example 
2. Open the Serial Monitor

**Expected Result:**
Get new value every 7 seconds.
https://github.com/Infineon/arduino-pas-co2-sensor/blob/bcd5899b9766797657f1f7ba675d8cf18d7f73d7/examples/serial-periodic/serial-periodic.ino#L12 https://github.com/Infineon/arduino-pas-co2-sensor/blob/bcd5899b9766797657f1f7ba675d8cf18d7f73d7/examples/serial-periodic/serial-periodic.ino#L56

**Actual Result:**
Serial Monitor Result:
```
11:50:48.714 -> j⸮⸮⸮⸮Υ(⸮⸮⸮o`⸮⸮1ġ⸮⸮serial initialized
11:50:56.426 -> get co2 error: 7
11:50:56.426 -> co2 ppm value : 0
11:51:03.382 -> get co2 error: 7
11:51:03.428 -> co2 ppm value : 0
11:51:10.397 -> get co2 error: 7
11:51:10.444 -> co2 ppm value : 0
11:51:17.442 -> get co2 error: 7
11:51:17.442 -> co2 ppm value : 0
11:51:24.467 -> co2 ppm value : 2179
11:51:31.434 -> get co2 error: 7
11:51:31.481 -> co2 ppm value : 0
11:51:38.454 -> get co2 error: 7
11:51:38.500 -> co2 ppm value : 0
11:51:45.480 -> get co2 error: 7
11:51:45.480 -> co2 ppm value : 0
11:51:52.457 -> get co2 error: 7
11:51:52.503 -> co2 ppm value : 0
11:51:59.463 -> get co2 error: 7
11:51:59.510 -> co2 ppm value : 0
11:52:06.495 -> get co2 error: 7
11:52:06.542 -> co2 ppm value : 0
11:52:13.527 -> get co2 error: 7
11:52:13.527 -> co2 ppm value : 0
11:52:20.504 -> get co2 error: 7
11:52:20.550 -> co2 ppm value : 0
11:52:27.525 -> co2 ppm value : 2189
11:52:34.552 -> get co2 error: 7
11:52:34.598 -> co2 ppm value : 0
11:52:41.553 -> get co2 error: 7
11:52:41.600 -> co2 ppm value : 0
11:52:48.605 -> get co2 error: 7
11:52:48.605 -> co2 ppm value : 0
11:52:55.575 -> get co2 error: 7
11:52:55.623 -> co2 ppm value : 0
11:53:02.600 -> get co2 error: 7
11:53:02.646 -> co2 ppm value : 0
11:53:09.633 -> get co2 error: 7
11:53:09.633 -> co2 ppm value : 0
11:53:16.610 -> get co2 error: 7
11:53:16.658 -> co2 ppm value : 0
11:53:23.635 -> co2 ppm value : 2191
11:53:30.622 -> get co2 error: 7
11:53:30.669 -> co2 ppm value : 0
```

**Frequency (1/1, 1/50, 1/100):**
1/1

**Build/Commit:**
v2.0.0

**Target:**
TTGO LoRa32 OLED, [EVAL_PASCO2_MINIBOARD](https://www.infineon.com/cms/de/product/evaluation-boards/eval_pasco2_miniboard/)

**Host OS and Version:**
Windows 10

**Environment**
Arduino IDE 1.8.13

###### THANKS!

jaenrig-ifx, 2022-02-28:
Hi @Lu-Ma-Ni,

Can you add a bit more delay before reading to the delay()? 1 sec (+1000) or so? 
https://github.com/Infineon/arduino-pas-co2-sensor/blob/bcd5899b9766797657f1f7ba675d8cf18d7f73d7/examples/serial-periodic/serial-periodic.ino#L67

When you do so, the reading attempts before getting a measure are always the same? 

Also can you run the serial-device-id example and say which device version do yo have?

Thanks!

Lu-Ma-Ni, 2022-02-28:
Thanks @jaenrig-ifx for the quick support!

I have added 1 second at line 67, but unfortunately this does not change anything.
```
14:27:36.756 -> !⸮⸮⸮serial initialized
14:27:56.961 -> get co2 error: 7
14:27:57.007 -> co2 ppm value : 0
14:28:04.983 -> get co2 error: 7
14:28:05.029 -> co2 ppm value : 0
14:28:13.318 -> get co2 error: 7
14:28:13.318 -> co2 ppm value : 0
14:28:21.330 -> get co2 error: 7
14:28:21.330 -> co2 ppm value : 0
14:28:29.550 -> get co2 error: 7
14:28:29.596 -> co2 ppm value : 0
14:28:37.556 -> get co2 error: 7
14:28:37.603 -> co2 ppm value : 0
14:28:45.885 -> get co2 error: 7
14:28:45.885 -> co2 ppm value : 0
14:28:53.886 -> co2 ppm value : 1981
14:29:02.170 -> get co2 error: 7
14:29:02.219 -> co2 ppm value : 0
14:29:10.196 -> get co2 error: 7
14:29:10.196 -> co2 ppm value : 0
14:29:18.495 -> get co2 error: 7
14:29:18.495 -> co2 ppm value : 0
14:29:26.490 -> get co2 error: 7
14:29:26.490 -> co2 ppm value : 0
14:29:34.783 -> get co2 error: 7
14:29:34.783 -> co2 ppm value : 0
14:29:42.786 -> get co2 error: 7
14:29:42.786 -> co2 ppm value : 0
14:29:51.045 -> co2 ppm value : 1975
14:29:59.077 -> get co2 error: 7
14:29:59.123 -> co2 ppm value : 0
14:30:07.353 -> get co2 error: 7
14:30:07.399 -> co2 ppm value : 0
14:30:15.357 -> get co2 error: 7
14:30:15.402 -> co2 ppm value : 0
14:30:23.642 -> get co2 error: 7
14:30:23.689 -> co2 ppm value : 0
14:30:31.667 -> get co2 error: 7
14:30:31.713 -> co2 ppm value : 0
14:30:39.961 -> get co2 error: 7
14:30:40.008 -> co2 ppm value : 0
14:30:47.960 -> co2 ppm value : 1991
14:30:56.243 -> get co2 error: 7
14:30:56.288 -> co2 ppm value : 0
```

**Serial Monitor of example serial-device-id**
```
14:22:09.320 -> ⸮]0⸮⸮⸮T⸮⸮u⸮h⸮z⸮}⸮⸮⸮⸮serial initialized
14:22:12.346 -> product id  : 2
14:22:12.346 -> revision id : 10
```

jaenrig-ifx, 2022-02-28:
Thanks @Lu-Ma-Ni,

Let us check with the Arduino Uno board and the same sensor version to see if the error can be reproduced.

Lu-Ma-Ni, 2022-02-28:
I just tried it on the Arduino Uno. Here the sensor works correctly. It is then probably because of the TTGO or because I had several devices connected to the I2C. If I find out something new, I post it here.

jaenrig-ifx, 2022-02-28:
That could be the case. Maybe the i2c voltage levels are different for each device? The co2 sensor is running at 3.3.V.

jaenrig-ifx, 2022-04-13:
Hi @Lu-Ma-Ni , 

The new release https://github.com/Infineon/arduino-pas-co2-sensor/releases/tag/v2.1.0 might solve the existing issues.

closed 2022-04-29 by jaenrig-ifx