Infineon/arduino-pas-co2-sensor #6: setPressRef() doesn't work, getting negative ppm

schumar, 2022-03-13:
### Prerequisites

_Can you reproduce the problem reliably?_  
**yes**

_Did you check current release notes for known issues?_  
Where can I find those?

_If this is not the latest release, have you checked newer releases?_
Using **2.0.0**

### Description

`setPressRef()` returns `1`; `getCO2()` returns negative values

### Steps to Reproduce

1. upload example `serial-periodic`
2. run it

### Expected Result

```
serial initialized
co2 ppm value : 400
co2 ppm value : 410
```

### Actual Result

```
serial initialized
get co2 error: 1
co2 ppm value : 0
pressure reference error: 1
co2 ppm value : -3325
get co2 error: 1
co2 ppm value : 0
pressure reference error: 1
co2 ppm value : -3329
get co2 error: 1
co2 ppm value : 0
pressure reference error: 1
co2 ppm value : -3333
get co2 error: 1
co2 ppm value : 0
pressure reference error: 1
co2 ppm value : -3335
get co2 error: 1
co2 ppm value : 0
pressure reference error: 1
```

### Frequency (1/1, 1/50, 1/100)

`serial_periodic` has
```c
#define PERIODIC_MEAS_INTERVAL_IN_SECONDS  7 
```

### Build/Commit:

_Please specify the release, like 1.4.1. If there is no release, give us the commit for the code you used._
**2.0.0**

### Target:

STM32F103 BluePill, PAS CO2 Miniboard

### Host OS and Version:

Debian GNU/Linux 11

### Compiler:

`gcc/10.3.1-2.3`

jaenrig-ifx, 2022-03-14:
Hi @schumar,

The output is reporting communication error. Can you share your wiring diagram? Are you using i2c? Are you using pull-up resistors?

jaenrig-ifx, 2022-03-15:
Mmm, looks good. Are the logic levels of the I2C ports working at 3.3V or 5V for this board?

schumar, 2022-03-15:
3.3V; I've lowered the I2C-frequency to 100kHz.

[image]

It looks like the Miniboard doesn't pull SDA fully to GND? (it's been a while since I bit-banged I2C, so maybe this is how it's supposed to look like):
[image]

jaenrig-ifx, 2022-03-16:
Does it work with 100KHz?
Definitively, that does not look very neat... Is it properly grounded with the osci? 

It would be good to know the sensor version, which can be obtained by running the serial-device-id example. But it won´t likely work with this board given the i2c issue.
Would you have by chance a xmc2go eval kit to check on that platform? Or an Arduino Uno?

schumar, 2022-03-16:
> Does it work with 100KHz?

no, same behaviour

> Definitively, that does not look very neat... Is it properly grounded with the osci?

Yes, double-checked just now

> It would be good to know the sensor version, which can be obtained by running the serial-device-id example. But it won´t likely work with this board given the i2c issue.

This works: (tried at 100kHz)
```
product id  : 2
revision id : 15
```

> Would you have by chance a xmc2go eval kit to check on that platform? Or an Arduino Uno?

XMC 2Go, Arduino Nano, even a TTGO ESP8266 -- I'll give them a try.

schumar, 2022-03-16:
Same with Arduino Nano :(

```
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3041
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3048
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3048
```

schumar, 2022-03-16:
And the same with an ESP8266:
```
serial initialized
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3037
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3045
get co2 error: 1
co2 ppm value : 0
co2 ppm value : -3047
```

schumar, 2022-03-16:
p.s. I commented out the `setPressRef()` calls, just in case you are wondering about the missing `pressure reference error: 1`

jaenrig-ifx, 2022-03-17:
And the same with the XMC 2Go? 
We will need for some time to reproduce the error by our application engineers. We will get back to you as soon as possible.

jaenrig-ifx, 2022-03-22:
Hi @schumar,

Did you have the chance to try with the XMC2Go?
There is a chance that the sensor is damaged, since negative values in the concentration could be due to the 12V supply not connected or not working.

schumar, 2022-03-22:
Hi @jaenrig-ifx , sorry, didn't have a lot of time in the last few days.

I have rebuilt the whole circuit (with the BluePill) on a new bread-board, and the SDA-at-1V issue went away (guess some grounding issue at the sensor side?).
Same behavior as before though (negative ppms).

I then switched to the XMC2Go, and that wasn't able to communicate with the sensor *at all*. I saw it sending via I2C, but the sensor just didn't respond.
Switched back to BluePill, and it worked again (with negative ppms).

I then came to the same conclusion as you -- sensor might be broken. Tried another one, instantly got sane values (around 700ppm). Didn't have time to switch back to the XMC2Go, but as it now works using the BluePill, and the XMC2Go isn't a good fit for the sensor anyway (it doesn't bring out V_USB to a dedicated pin, but I need those sweet 5V to power a boost-converter, so I can get the 12V required), I'll stick with the BluePill.

Thanks for your help! I'll ping you tomorrow via WebEx, in case someone wants to have a look at the broken sensor (and yes, I assume that it was perfectly fine before I somehow managed to destroy it.)

closed 2022-03-22 by schumar